matthias-bs/ATC_MiThermometer #4: Example code reports one device more than expected

Art-ut-Kia, 2023-01-26:
**Describe the bug**
I ran the code as is with 3 mijia sensors LYWSD03MMC and with their MAC's listed in knownBLEAdresses vector.
In the serial monitor, it displayed:
            BLE Devices found (total): 4

**To Reproduce**
as per the bug description
4. See error

**Expected behavior**
should display:
            BLE Devices found (total): 3

**Embedded Device (please complete the following information):**
 - Board: Wemos D1 R32
 - IDE 2.0.3
 - Library Version: latest
 - Pinning/Wiring: not applicable since everything is wireless :))

**Additional context**
As far as I know, all devices around (smartphone, ...) have their bluetooth off.

matthias-bs, 2023-01-26:
It's not a bug, it's a feature! ;-)

I had the same experience with https://github.com/theengs/decoder/blob/development/examples/ESP32/ScanAndDecode/ScanAndDecode.ino - `MyAdvertisedDeviceCallbacks` were triggered multiple times for the same sensor; this makes keeping track of already received  sensors difficult. 

Setting `ScanFilterMode` to `CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE` seems to restrict callback invocation to once per device.

So you can replace
```
    NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DEVICE);
```
by
```
    NimBLEDevice::setScanFilterMode(CONFIG_BTDM_SCAN_DUPL_TYPE_DATA_DEVICE);
```

If you prefer to get updates of sensors already received during the scan cycle, you have to implement a more advanced counting yourself.

Art-ut-Kia, 2023-01-26:
Understood.
Thanks for the explanations.

closed 2023-01-27 by Art-ut-Kia