matthias-bs/ATC_MiThermometer #5: Getting incorrect readings

OriginalCREart, 2023-02-20:
Hi,
I have flashed the Mi Thermometer with this repo [https://github.com/atc1441/ATC_MiThermometer](https://github.com/atc1441/ATC_MiThermometer) and this is the [flashed bin file](https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer.bin).

I grap an ESP32-wroom-32d board, I have used your [Arduino sketch](https://github.com/matthias-bs/ATC_MiThermometer/blob/main/examples/ATC_MiThermometer_Nonblocking/ATC_MiThermometer_Nonblocking.ino) and upload it to the ESP. everything seems to work fine from scanning to identifying the Mi Thermometers, but the output of the reading are not correct, sample output
ensor #0 -> temperature: -92.16C ; humidity: 241.26% ; voltage: 60.427V ; battery charge: 179% ; RF power: -75dBm
sensor #1 -> temperature: -156.16C ; humidity: 233.62% ; voltage: 53.771V ; battery charge: 247% ; RF power: -86dBm
sensor #2 -> temperature: -156.16C ; humidity: 215.70% ; voltage: 39.435V ; battery charge: 191% ; RF power: -85dBm

Only the RF power seems correct!  what I did wrong?

matthias-bs, 2023-02-20:
Hi,

this is strange... There are different encodings depending on the selected Mi Thermometer firmware, but I thought I fixed this already (as mentioned in the issue you are referring to at the end). Maybe there is a third variant I did not consider yet...

Can you please confirm that you installed the library from https://github.com/matthias-bs/ATC_MiThermometer (and not from https://github.com/matthias-bs/ESP32_ATC_MiThermometer_Library)?

If yes, could you please compile again with debugging enabled (by setting "Core Debug Level: Debug" in the Arduino IDE) and provide some output?

Did you already try the other example (ATC_MiThermometer_Client)?

Regards
Matthias

OriginalCREart, 2023-02-20:
Hi, 

Yes I can confirm that I have used the library from [https://github.com/matthias-bs/ATC_MiThermometer](https://github.com/matthias-bs/ATC_MiThermometer) and **not** from [https://github.com/matthias-bs/ESP32_ATC_MiThermometer_Library](https://github.com/matthias-bs/ESP32_ATC_MiThermometer_Library), see the attached screenshot (green one). 

[image]

Regarding "**ATC_MiThermometer_Client**", here are the results I got, still the same :confused:

Sensor 0: A4:C1:38:XX:XX:XX
-87.04°C
218.22%
40.459V
8%
-43dBm

Sensor 1: A4:C1:38:XX:XX:XX
-179.20°C
233.64%
53.003V
26%
-90dBm

Sensor 2: A4:C1:38:XX:XX:XX
-135.68°C
243.83%
62.731V
230%
-84dBm

BLE Devices found (total): 20
Iteration 0 - Free heap is 220844

OriginalCREart, 2023-02-20:
I think it is now fixed, I closed the Arduino IDE, and reopened it again, it shows updates notification, one of them was the library "**ATC_MiThermometer**" version **0.0.2** instead of **0.1.0** as seen in the screenshot. 

[image]

### After update:
Data Readings (using: [ATC_MiThermometer_Client.ino](https://github.com/matthias-bs/ATC_MiThermometer/blob/main/examples/ATC_MiThermometer_Client/ATC_MiThermometer_Client.ino))

Sensor 0: A4:C1:38:XX:XX:XX
22.90°C
70.00%
3.000V
88%
-68dBm

Sensor 1: A4:C1:38:XX:XX:XX
18.50°C
68.00%
3.023V
91%
-87dBm

Sensor 2: A4:C1:38:XX:XX:XX
20.20°C
64.00%

3.061V
95%
-76dBm

BLE Devices found (total): 17
Iteration 1 - Free heap is 230096

Data Readings (using: [ATC_MiThermometer_Nonblocking.ino](https://github.com/matthias-bs/ATC_MiThermometer/blob/main/examples/ATC_MiThermometer_Nonblocking/ATC_MiThermometer_Nonblocking.ino)

sensor # 0 -> temperature: 22.30C ; humidity: 59.00% ; voltage: 2.963V ; battery charge: 84% ; RF power: -70dBm
sensor # 1 -> temperature: 18.50C ; humidity: 67.00% ; voltage: 3.022V ; battery charge: 90% ; RF power: -92dBm
sensor # 2 -> temperature: 20.10C ; humidity: 64.00% ; voltage: 3.057V ; battery charge: 95% ; RF power: -78dBm

## Thanks you so much for your reply and your time. I hope this will help others too.

closed 2023-02-20 by OriginalCREart

matthias-bs, 2023-02-21:
You are welcome!

Besides, I noticed that there is still a comment referring to the wrong repository... I will make a new release now.