matthias-bs/ATC_MiThermometer #6: miThermometer.clearScanResults(); Don't seem to clear anything

johnostr, 2023-08-26:
I was thinking that after miThermometer.clearScanResults(); or miThermometer.resetData(); ran the BLE data would clear.  In my case when I run the serial.printf code again (a second time) after the miThermometer.clearScanResults(), just like in the client example, the data is still displaying.  Same data shows again.  

What is the miThermometer.clearScanResults() and miThermometer.resetData() suppose to do?

Other than that the code works great !!

matthias-bs, 2023-08-26:
miThermometer.resetData() sets the valid flag to false.
miThermometer.clearScanResults() clears the BLE scan data to free allocated memory, but leaves the acquired sensor data untouched.

johnostr, 2023-08-27:
Many thanks, great to know what they do.

closed 2023-08-27 by johnostr