matthias-bs/ATC_MiThermometer #7: Add Reed Switch Support (pvvx custom firmware)

chinswain, 2024-04-02:
Would it be possible to add support for reading the state of the reed switch that's been added the custom format?

I have 10 sensors located in 10 beehives (they fix perfectly inside) that are currently reporting temp and humidity (Healthy beehives are always around 35C) which are monitored by an ESP32 that sends the data via LoRa to my house a few miles away.

I'd like to place a magnet in the lid to activate the reed switch I've added to the LYWSD03MMC's so I have an alert if anyone opens the hives.

Custom format (all data little-endian):
```
uint8_t     size;   // = 18
uint8_t     uid;    // = 0x16, 16-bit UUID
uint16_t    UUID;   // = 0x181A, GATT Service 0x181A Environmental Sensing
uint8_t     MAC[6]; // [0] - lo, .. [6] - hi digits
int16_t     temperature;    // x 0.01 degree
uint16_t    humidity;       // x 0.01 %
uint16_t    battery_mv;     // mV
uint8_t     battery_level;  // 0..100 %
uint8_t     counter;        // measurement count
uint8_t     flags;  // GPIO_TRG pin (marking "reset" on circuit board) flags: 
                    // bit0: Reed Switch, input
                    // bit1: GPIO_TRG pin output value (pull Up/Down)
                    // bit2: Output GPIO_TRG pin is controlled according to the set parameters
                    // bit3: Temperature trigger event
                    // bit4: Humidity trigger event
```

matthias-bs, 2024-04-03:
Hi,

you have created an interesting application! I just found the information regarding the counter/flags here:
https://github.com/pvvx/ATC_MiThermometer?tab=readme-ov-file#custom-format-all-data-little-endian

If you like, you can implement the changes and create a pull request. Otherwise it may take a while until I find some to do this. 

Best regards,
Matthias

chinswain, 2024-04-03:
Hi Matthias,

It's an excellent solution and at a very low cost compared to commercial solutions! I know exactly how each beehive is doing in the middle of winter so can go and check for problems.

I think I've worked out how to extract the reed switch status, I can't find the count though (position 13 seems to increment on each advertisement) 

https://github.com/chinswain/ATC_MiThermometer/blob/main/src/ATC_MiThermometer.cpp

Added to .h
```
uint8_t             count;        	//!< measurement count
bool 		reedSwitchState;
bool 		gpioTrgOutput;
bool 		controlParameters;
bool 		tempTriggerEvent;
bool 		humiTriggerEvent;

```

Added to .cpp
```
// Count
uint8_t flagsByte = foundDevices.getDevice(i).getServiceData().c_str()[14];

// Flags
data[n].reedSwitchState = flagsByte & 0x01; // Extract bit 0 (Reed Switch)
data[n].gpioTrgOutput = (flagsByte >> 1) & 0x01; // Extract bit 1 (GPIO_TRG pin output)
data[n].controlParameters = (flagsByte >> 2) & 0x01; // Extract bit 2 (Control parameters)
data[n].tempTriggerEvent = (flagsByte >> 3) & 0x01; // Extract bit 3 (Temperature trigger event)
data[n].humiTriggerEvent = (flagsByte >> 4) & 0x01; // Extract bit 4 (Humidity trigger event)

```

Data from the flashing tool shows the counter and flag data is in the advertisement so I assume the device is configured and transmitting correctly:

Vbat: 3002 mV , Temp: 23.30°C, Humi: 48.06%, ID: 1464, Reed switch counter: 11, flg: 0x05:r1/t0

Output from library:
Sensor 1: a4:c1:38:9f:03:47
23.90°C
47.59%
2.971V
100%
-44dBm
Count: 63
Reed Switch State: Closed
GPIO TRG Output: Low
Control Parameters: Set
Temperature Trigger Event: Not Triggered
Humidity Trigger Event: Not Triggered

matthias-bs, 2024-04-03:
Maybe you can configure if the counter is incremented on each advertisement or only on events?

You can create a pull request by clicking "Contribute" in your branch.

matthias-bs, 2024-04-07:
@chinswain Added in https://github.com/matthias-bs/ATC_MiThermometer/pull/8. Thanks for your contribution.
Released [ATC_MiThermometer v0.3.1](https://github.com/matthias-bs/ATC_MiThermometer/releases/tag/v0.3.1).

closed 2024-04-07 by matthias-bs