matthias-bs/ATC_MiThermometer #9: GetName()

kharar, 2024-04-19:
**Is your feature request related to a problem? Please describe.**
When fetchins sensor data for several devices I would also like to include the bluetooth name of the sensor in my readout. Currently all we got is the MAC and I don't want to maintain a list copy of the BTnames in code.

**Describe the solution you'd like**
Update the struct MiThData_S and supporting code to hold the string value of the bluetooth name:

// MiThermometer data struct / type
struct MiThData_S {
        bool        valid;          //!< data valid
        int16_t     temperature;    //!< temperature x 100°C
        uint16_t    humidity;       //!< humidity x 100%
        uint16_t    batt_voltage;   //!< battery voltage [mv]
        uint8_t     batt_level;     //!< battery level   [%]
        int16_t     rssi;           //!< RSSI [dBm]
        uint8_t     count;        	//!< measurement count
        std::string name;           //!< bluetooth name
        bool 	    reedSwitchState;
        bool 	    gpioTrgOutput;
        bool 	    controlParameters;
        bool 	    tempTriggerEvent;
        bool 	    humiTriggerEvent;
};

In such a way that we will be able to get the BT name just like any other sensor detail, for instance:
std::string myName = miThDat.name;
and
int16_t myRssi = miThDat.rssi

**Describe alternatives you've considered**
I have already tried modifying the supporting code myself, but I got to admit that I am in way over my head

I got as far as to insert these two lines at line 168 of ATC_MiThermometer.cpp, right above the data[n].rssi code:

                // Received Bluetooth Name
                data[n].name = foundDevices.getDevice(i).getName().c_str();

but this is not enough to get it working

**Additional context**
For someone who is not a complete noob, this is probably an easy fix :-)

matthias-bs, 2024-04-25:
Hi,

I started looking into this right now, but most of the devices within reach to not seem to provide a name.
Apart from that, you already have the implementation in your request.

Regards,
Matthias

matthias-bs, 2024-04-25:
Added in https://github.com/matthias-bs/ATC_MiThermometer/pull/10

kharar, 2024-04-25:
It is possible that you don't use it but all bluetooth devices support advertising this information. For the LYWSD03 you can even edit the name in the Telink Flasher:
[image]

To my regret the implementation I came up with is not working. I believe it has something to do with the data structure of NimBLE, unfortunately I am not code wizard enough to figure out how to pull this string from within your framework

matthias-bs, 2024-04-25:
Ah, o.k. I just added it and created a new release v0.4.1 which should be available in the Arduino library manager within an hour. Hope it works - please let me know.

kharar, 2024-04-25:
Thanks, I just pulled it from here and tested the example code from the from page but unfortunately the name is not passed. My output looks like this:
```
Sensor 0: A4:C1:38:B2:90:45
Name: 
23.68°C
33.08%
3.938V
100%
-90dBm

```

I was expecting:

```
Sensor 0: A4:C1:38:B2:90:45
Name: Bathroom
23.68°C
33.08%
3.938V
100%
-90dBm

```

kharar, 2024-04-25:
Oh snap! That was probably version 0.3.1 I just pulled...
I'll wait a little longer before I write again :)

kharar, 2024-04-25:
As far as I figured out, if I replace

`            data[n].name = foundDevices.getDevice(i).getName().c_str();
`
with
`            data[n].name = "Test"; //foundDevices.getDevice(i).getName().c_str();
`
Then the name "Test" is passed correctly, therefore I am assuming that the fault lies somewhere along the `foundDevices.getDevice(i).getName().c_str()` but as I mentioned before my understanding of the NimBLE integration is being challenged a lot here.

matthias-bs, 2024-04-26:
I found that it works (most of the time) if you change to active scan mode in `ATC_MiThermometer.cpp`:

```
_pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
```

I don't know if this expected behaviour or a bug. And sometimes the name is still missing.

You can set the Debug Level to "Debug" in the Arduino IDE and recompile to get more info from NimBLE and this library.

matthias-bs, 2024-04-26:
See https://github.com/matthias-bs/ATC_MiThermometer/pull/11 and [Release v0.4.2](https://github.com/matthias-bs/ATC_MiThermometer/releases/tag/v0.4.2)

kharar, 2024-04-27:
With 0.4.2 I can confirm that it works as you described, also accasionally missing a name now and then, however changing scan time from 5 to 10 does not seem to affect this.
It seems safe enough to assume this particular sub-problem lies within the mechanics of NimBLE or even the Bluetooth protocol itself.

Thank you so much for your effort, I am all fine with closing this issue here.

matthias-bs, 2024-04-27:
I'll leave it is at is for now. I used 15 seconds when I got the name reliably. Maybe it is related to some multiple of the advertising cycle time.

closed 2024-04-27 by matthias-bs

matthias-bs, 2024-04-27:
Here is the answer: https://development.libelium.com/ble-networking-guide/scanning-ble-devices

_On the other hand, in active scanning the module will request more information once an advertisement is received, and the advertiser will answer with information like friendly name and supported profiles._ 

I think the reason for missing names during active scanning is that I stop scanning when I receive an advertisement with the wanted address. At that time, the response to the advertising request may not have been received yet.