matthias-bs/ATC_MiThermometer #12: Deinitialize BLE Not working: Failed with error: 259

chinswain, 2024-06-16:
Any idea on this one? I need to switch to WiFi periodically to transmit the data back but can't deinit.
If I call ```NimBLEDevice::deinit(true);``` after clearing the scan results I get:

```E NimBLEDevice: esp_nimble_hci_and_controller_deinit() failed with error: 259```

```
#include "ATC_MiThermometer.h"

const int scanTime = 5;  // BLE scan time in seconds

// List of known sensors' BLE addresses
std::vector<std::string> knownBLEAddresses = { "A4:C1:38:C5:3E:18" };

ATC_MiThermometer miThermometer(knownBLEAddresses);

void setup() {
  Serial.begin(115200);

  // Initialization
  miThermometer.begin();

}

void loop() {
  miThermometer.begin();

  // Set sensor data invalid
  miThermometer.resetData();

  // Get sensor data - run BLE scan for <scanTime>
  unsigned found = miThermometer.getData(scanTime);

  for (int i = 0; i < miThermometer.data.size(); i++) {
    if (miThermometer.data[i].valid) {
      Serial.println();
      Serial.printf("Sensor %d: %s\n", i, knownBLEAddresses[i].c_str());
      Serial.printf("Name: %s\n", miThermometer.data[i].name.c_str());
      Serial.printf("%.2f°C\n", miThermometer.data[i].temperature / 100.0);
      Serial.printf("%.2f%%\n", miThermometer.data[i].humidity / 100.0);
      Serial.printf("%.3fV\n", miThermometer.data[i].batt_voltage / 1000.0);
      Serial.printf("%d%%\n", miThermometer.data[i].batt_level);
      Serial.printf("%ddBm\n", miThermometer.data[i].rssi);
      Serial.printf("Measurement Count: %d\n", miThermometer.data[i].count);
      // Print boolean flags
      Serial.printf("Reed Switch State: %s\n", miThermometer.data[i].reedSwitchState ? "Open" : "Closed");
      Serial.printf("GPIO TRG Output: %s\n", miThermometer.data[i].gpioTrgOutput ? "High" : "Low");
      Serial.printf("Control Parameters: %s\n", miThermometer.data[i].controlParameters ? "Set" : "Not Set");
      Serial.printf("Temperature Trigger Event: %s\n", miThermometer.data[i].tempTriggerEvent ? "Triggered" : "Not Triggered");
      Serial.printf("Humidity Trigger Event: %s\n", miThermometer.data[i].humiTriggerEvent ? "Triggered" : "Not Triggered");
      Serial.println();
    }
  }
  Serial.println("BLE Devices found (total): " + String(found));

  // Delete results from BLEScan buffer to release memory
  miThermometer.clearScanResults();

  delay(5000);
  NimBLEDevice::deinit(true);
}
```

gcharles81, 2025-01-06:
Hello Did you solve the issue ? thanks

chinswain, 2025-01-06:
It looks like it's just been made live, here's the migration guide: https://github.com/h2zero/NimBLE-Arduino/releases/tag/2.1.3

gcharles81, 2025-01-07:
seems cannot compile using the client Example ,

using : 
lib ATC_MiThermometer  version=0.4.2
lib NimBLE-Arduino  version=2.1.3

compilation error on platformio - lib/ATC_MiThermometer/src/ATC_MiThermometer.cpp:114:53: error: request for member 'getServiceData' in 'foundDevices.NimBLEScanResults::getDevice(i)', which is of pointer type 'const NimBLEAdvertisedDevice*' (maybe you meant to use '->' ?)

using below example included in the library  

#include "ATC_MiThermometer.h"

const int scanTime = 5; // BLE scan time in seconds

// List of known sensors' BLE addresses
std::vector<std::string> knownBLEAddresses = {"a4:c1:38:b8:1f:7f", "a4:c1:38:bf:e1:bc"};

ATC_MiThermometer miThermometer(knownBLEAddresses);

// Iteration counter
int iteration = 0;

void setup() {
    Serial.begin(115200);
    
    // Print free heap before initialization
    Serial.println("Starting:    Free heap is " + String(ESP.getFreeHeap()));
    
    // Initialization
    miThermometer.begin();
    
    // Print free heap after initialization
    Serial.println("Initialized: Free heap is " + String(ESP.getFreeHeap()));
}

void loop() {
    // Set sensor data invalid
    miThermometer.resetData();
    
    // Get sensor data - run BLE scan for <scanTime>
    unsigned found = miThermometer.getData(scanTime);

    for (int i=0; i < miThermometer.data.size(); i++) {  
        if (miThermometer.data[i].valid) {
            Serial.println();
            Serial.printf("Sensor %d: %s\n", i, knownBLEAddresses[i].c_str());
            Serial.printf("Name: %s\n", miThermometer.data[i].name.c_str());
            Serial.printf("%.2f°C\n", miThermometer.data[i].temperature/100.0);
            Serial.printf("%.2f%%\n", miThermometer.data[i].humidity/100.0);
            Serial.printf("%.3fV\n",  miThermometer.data[i].batt_voltage/1000.0);
            Serial.printf("%d%%\n",   miThermometer.data[i].batt_level);
            Serial.printf("%ddBm\n",  miThermometer.data[i].rssi);
            Serial.printf("Measurement Count: %d\n", miThermometer.data[i].count);
            // Print boolean flags
            Serial.printf("Reed Switch State: %s\n", miThermometer.data[i].reedSwitchState ? "Open" : "Closed");
            Serial.printf("GPIO TRG Output: %s\n", miThermometer.data[i].gpioTrgOutput ? "High" : "Low");
            Serial.printf("Control Parameters: %s\n", miThermometer.data[i].controlParameters ? "Set" : "Not Set");
            Serial.printf("Temperature Trigger Event: %s\n", miThermometer.data[i].tempTriggerEvent ? "Triggered" : "Not Triggered");
            Serial.printf("Humidity Trigger Event: %s\n", miThermometer.data[i].humiTriggerEvent ? "Triggered" : "Not Triggered");
            Serial.println();
         }
    }
    Serial.println("BLE Devices found (total): " + String(found));

    // Delete results from BLEScan buffer to release memory
    miThermometer.clearScanResults();
    
    // Print iteration counter and free heap
    Serial.println("Iteration " + String(iteration++) + " - Free heap is " + String(ESP.getFreeHeap()));
    Serial.println("---");

    delay(5000);
}
*/

matthias-bs, 2025-01-11:
Hi,

I already made an attempt to port some of my other BLE code to the new NimBLE library, but this was not working immediately: https://github.com/h2zero/NimBLE-Arduino/issues/826

matthias-bs, 2025-01-25:
@chinswain and @gcharles81 Hi, I have updated to NimBLE v2.2.0. Please try again.

gcharles81, 2025-01-26:
> [@chinswain](https://github.com/chinswain) and [@gcharles81](https://github.com/gcharles81) Hi, I have updated to NimBLE v2.2.0. Please try again.

Thanks @matthias-bs  but in my case my code is also using "BLEAdvertisingReader.h" and it will not compile with latest changes also .

matthias-bs, 2025-02-03:
Closed due to inactivity.

closed 2025-02-03 by matthias-bs