matthias-bs/ATC_MiThermometer #13: Not working on Arduino Nano ESP32

chinswain, 2024-07-08:
I've just tried setting up the library on an Arduino Nano ESP32 clone but I don't get any sensor data returned, for some reason it's unable to find the thermostats. the output is:

```
BLE Devices found (total): 8
Iteration 25 - Free heap is 271512
---
BLE Devices found (total): 8
Iteration 26 - Free heap is 271512
---
BLE Devices found (total): 9
Iteration 27 - Free heap is 271512
```

Debug:

```
Advertised Device: Name: , Address: 0c:4b:c9:77:57:3f, manufacturer data: 0600010920220552d8ef8693a70f579ac45628f29cf49101d24c460b5aAdvertised Device: Name: , Address: 5e:ed:34:eb:f0:ad, manufacturer data: e00056dcca6606d3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data: Advertised Device: Name: , Address: 67:0b:e3:42:11:4b, manufacturer data: e000567bca6cd7a5, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data: Advertised Device: Name: MJ_HT_V1, Address: 58:2d:34:37:34:00, serviceUUID: 0x180f
Service Data:
UUID: 0xfe95, Data: P ��Advertised Device: Name: , Address: 45:c0:cd:11:2f:00, serviceUUID: 0xfef3Advertised Device: Name: , Address: 54:48:e6:ba:45:b7, manufacturer data: 8f030a10030b00b545bae6485481Advertised Device: Name: , Address: 6a:6a:6a:5c:c8:e3, serviceUUID: 0xfe9f
Service Data:
UUID: 0xfe9f, Data: Whitelist contains:Assigning scan results...haveName(): 0getName(): Found: 67:0b:e3:42:11:4b  comparing to: a4:c1:38:c5:3e:18
Found: 67:0b:e3:42:11:4b  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 0c:4b:c9:77:57:3f  comparing to: a4:c1:38:c5:3e:18
Found: 0c:4b:c9:77:57:3f  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 45:c0:cd:11:2f:00  comparing to: a4:c1:38:c5:3e:18
Found: 45:c0:cd:11:2f:00  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 5e:ed:34:eb:f0:ad  comparing to: a4:c1:38:c5:3e:18
Found: 5e:ed:34:eb:f0:ad  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 54:48:e6:ba:45:b7  comparing to: a4:c1:38:c5:3e:18
Found: 54:48:e6:ba:45:b7  comparing to: a4:c1:38:a2:2d:de
haveName(): 1getName(): MJ_HT_V1Found: 58:2d:34:37:34:00  comparing to: a4:c1:38:c5:3e:18
Found: 58:2d:34:37:34:00  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 6a:6a:6a:5c:c8:e3  comparing to: a4:c1:38:c5:3e:18
Found: 6a:6a:6a:5c:c8:e3  comparing to: a4:c1:38:a2:2d:de
BLE Devices found (total): 7
Iteration 9 - Free heap is 271512
---
```

For an old ESP32 32S it works as expected:

```
Found: 54:48:e6:ba:45:b7  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 6a:6a:6a:5c:c8:e3  comparing to: a4:c1:38:c5:3e:18
Found: 6a:6a:6a:5c:c8:e3  comparing to: a4:c1:38:a2:2d:de
haveName(): 1getName(): mi_mtkFound: 04:cf:8c:78:69:e4  comparing to: a4:c1:38:c5:3e:18
Found: 04:cf:8c:78:69:e4  comparing to: a4:c1:38:a2:2d:de
haveName(): 1getName(): ATC_C53E18Found: a4:c1:38:c5:3e:18  comparing to: a4:c1:38:c5:3e:18 -> Match! Index: 0Length of ServiceData: 15Custom formatFound: a4:c1:38:c5:3e:18  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 52:04:69:5c:4b:f3  comparing to: a4:c1:38:c5:3e:18
Found: 52:04:69:5c:4b:f3  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 5e:7f:6f:d5:c8:84  comparing to: a4:c1:38:c5:3e:18
Found: 5e:7f:6f:d5:c8:84  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 67:38:8f:91:8b:d4  comparing to: a4:c1:38:c5:3e:18
Found: 67:38:8f:91:8b:d4  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: 5d:9d:14:8e:ff:d3  comparing to: a4:c1:38:c5:3e:18
Found: 5d:9d:14:8e:ff:d3  comparing to: a4:c1:38:a2:2d:de
haveName(): 0getName(): Found: a4:c1:38:a2:2d:de  comparing to: a4:c1:38:c5:3e:18
Found: a4:c1:38:a2:2d:de  comparing to: a4:c1:38:a2:2d:de -> Match! Index: 1Length of ServiceData: 15Custom formathaveName(): 0getName(): Found: 44:54:27:54:71:0e  comparing to: a4:c1:38:c5:3e:18
Found: 44:54:27:54:71:0e  comparing to: a4:c1:38:a2:2d:de
```
 - Board: [esp32-s3-nano clone](https://www.waveshare.com/esp32-s3-nano.htm) 
 - ESP32-S3R8
 - Arduino Version: 2.2.1
 - Library Version 0.4.2
 - Using example code

chinswain, 2024-07-08:
I've tried a basic scan using nimBLE and Bluedroid, both cannot find the LYWSD03MMC - beware of this device if you want to use this library.

matthias-bs, 2024-07-08:
Maybe this is a signal strength issue, i.e. the board's BLE radio does not work as good as the one you used previously? It could be 
1. a hardware issue
2. a Bluetooth core SW issue
3. an ATC_MiThermometer issue.

You could try https://decoder.theengs.io/ to rule out 1.
Your test with NimBLE and BlueDroid seems to rule out 2.

Maybe you could also enable more debug messages from the BLE core?

chinswain, 2024-07-08:
Not picked up via theengs - original LYWSD03MMC is found.

Range seems good via the ceramic antenna, picking updates quite far away.

I've started a discussion on NimBLE to see if anyone has any ideas, I've ordered a generic ESP32-S3 dev board to test if it's an issue with the chip or specific to Arduinos implementation of it.

chinswain, 2025-01-18:
I ended up returning the nano and sticking with ESP32 S3.

closed 2025-01-18 by chinswain