matthias-bs/ATC_MiThermometer #15: Can't compile on ESP32

adrianAzoitei, 2025-01-18:
Hi,

I'm just getting started with ESP32 and I've got a Xiaomi BLE temperature & humidity sensor around that I wanted to connect to the board, so I ended up on this repository.

I'm trying to run the example, but I get the following compilation error from the Arduino IDE:
```bash
/Users/adrianazoitei/Documents/Arduino/libraries/ATC_MiThermometer/src/ATC_MiThermometer.cpp: In member function 'void ATC_MiThermometer::begin(bool)':
/Users/adrianazoitei/Documents/Arduino/libraries/ATC_MiThermometer/src/ATC_MiThermometer.cpp:83:16: error: 'class NimBLEScan' has no member named 'setAdvertisedDeviceCallbacks'
   83 |     _pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
      |                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/adrianazoitei/Documents/Arduino/libraries/ATC_MiThermometer/src/ATC_MiThermometer.cpp: In member function 'unsigned int ATC_MiThermometer::getData(uint32_t)':
/Users/adrianazoitei/Documents/Arduino/libraries/ATC_MiThermometer/src/ATC_MiThermometer.cpp:93:51: error: conversion from 'bool' to non-scalar type 'NimBLEScanResults' requested
   93 |     BLEScanResults foundDevices = _pBLEScan->start(duration, false /* is_continue */);
      |                                   ~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
....
```

I did some digging and saw that `NimBLEScan.setAdvertisedDeviceCallbacks` was indeed removed [here](https://github.com/h2zero/NimBLE-Arduino/commit/65e29190351aa01eb54ab813db3d99139bce61e8#diff-f45f03296a168c53918c90ba2a4a79569b69091c6afecb7735547f44ba116bc1R67) , so I tried downgrading NimBLE-Arduino to 1.3.0, but then I get another compilation error:
```bash
In file included from /Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLEDescriptor.h:23,
                 from /Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLE2904.h:23,
                 from /Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLE2904.cpp:25:
/Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLECharacteristic.h:104:32: error: 'time_t' has not been declared
  104 |     std::string       getValue(time_t *timestamp = nullptr);
      |                                ^~~~~~
/Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLECharacteristic.h:116:32: error: 'time_t' has not been declared
  116 |     T                 getValue(time_t *timestamp = nullptr, bool skipSizeCheck = false) {
      |                                ^~~~~~
/Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLECharacteristic.h:155:5: error: 'time_t' does not name a type
  155 |     time_t                         m_timestamp;
      |     ^~~~~~
/Users/adrianazoitei/Documents/Arduino/libraries/NimBLE-Arduino/src/NimBLECharacteristic.h:45:1: note: 'time_t' is defined in header '<ctime>'; did you forget to '#include <ctime>'?
  ....
```

Did anyone run into similar issues? Any guidance on how I can proceed?

chinswain, 2025-01-18:
It needs to be ported to work with the new nimBLE library, currently in progress and hold due to this issue: https://github.com/h2zero/NimBLE-Arduino/issues/826

adrianAzoitei, 2025-01-18:
@chinswain thanks for your reply! What's the last version of NimBLE-Arduino that is known to be compatible?

adrianAzoitei, 2025-01-18:
Ah, 1.4.3.

matthias-bs, 2025-01-25:
@adrianAzoitei Hi, I have updated to NimBLE v2.2.0. Please try again.

matthias-bs, 2025-02-03:
Closed due to inactivity.

closed 2025-02-03 by matthias-bs